rdawg-pidinst/schema #2: Problems exporting RST file to e.g. ODT or DOCX

DrMaggie, 2018-10-25:
Hi Rolf,
thanks for putting together the schema - much appreciated! 
However, I would really like to save the contents of the schema.rst file to a more PC editing-friendly format, in order to more easily share it with some (offline) colleagues. I've tried on-the-fly conversion from RST to e.g. DOCX, but the output looks horrible. (The table cell borders are reproduced as "|" and "-" characters, creating a weird "ascii-fied" appearance that won't convert to a proper table.)
Are you able to point me to a Word- or LibreOffice Writer-compatible version of your schema table, that comes out like a proper table?
/Maggie
PS While I can see that it may be very useful for RDA WGs to 1) use GitHub or similar platforms for file distribution in contrast to e.g. Google Drive that may be inaccessible for some people; and 2) not to provide their outputs in proprietary formats like DOCX, it does nevertheless often frustrate "common" users like myself who work in a Windows environment. I would therefore suggest to provide an ODT version in parallel to the RST one for this type of text document.

markusstocker, 2018-10-25:
For ODT this works reasonably well

```
pandoc -f rst -t odt schema.rst -o schema.odt
```
I have committed `schema.odt`.

The same for DOCX works as well but the table seems to be rendered without borders.

DrMaggie, 2018-10-25:
Thank you very much for this, Markus! 
I actually tried that same conversion with pandoc, but in my case both the DOCX and ODT outputs looked quite bad (as detailed in my issue report). I also tried out the on-line conversion tool CloudConvert (http://cloudconvert.com) with pretty much the same outcome. 
Obviously what we are discussing now is an _intermediate-level working document_, for which it is probably perfectly okay to assume that end users are willing (and able) to run a converter, if needed. However, any finalized reports and recommendations should "by default" also be made available in a parallel version. I note that this is sadly not the case for all RDA group _final_ outputs published up to the present...

markusstocker, 2018-10-25:
I am using pandoc version 1.19.2.4. Could be the reason for the differences.

DrMaggie, 2018-10-25:
Ah, yes: I used the most recent version for Windows (v2.3.1), just downloaded from https://pandoc.org/. (Newest isn't always best ;-) Anyway, I vote to close this issue!

closed 2018-10-25 by DrMaggie

RKrahl, 2018-10-25:
First of many thanks to @markusstocker for fixing this so quickly!

Although the issue is closed, I'd still like to reply to your comment, @DrMaggie : the questions of accessibility or closed versus open file formats was not the main reason for moving to GitHub. The work on the Google Doc was great to start the draft, because it helped to involve so many people very quickly. But in the end, I found it very difficult to keep track of all proposals for modifications and to get a clue, if we reached agreement on any of these proposals or not. If no more comments follow to some argument, does that mean that others did accept it? So I felt, we were at a point where a more formalized approach to deal with proposals for modification would help. And I suggest that discussions in GitHub issues is a very suitable tool to do that: we can open them, discuss them in a somewhat more focused manner, and formally conclude and close them.

I'm very happy to provide an ODT version in parallel to the RST. (I will be even happier if Markus helps to care about it.) But I suggest that we agree that the RST version (or any other ASCII based format) should be the master that all other formats are generated from. Simply because it's much easier to track changes in an ASCII based format.