rdawg-pidinst/schema #4: Description only recommended

louatbodc, 2018-10-26:
Slightly concerned the schema has moved away from the use-case alignment in some instances. For example, description was requested by all 10 use cases but here is only a recommended.

RKrahl, 2018-10-27:
If the property is suggested by use cases, it certainly should be included in the schema. If it is suggested by many use cases, this may be a hint to classify it as recommended. But I still did not understand why the number of use cases suggesting a property would be a criterion to classify it as mandatory.

louatbodc, 2018-10-29:
I guess my thoughts are that we need to justify why properties are in and out, recommended/mandatory etc. and this should be really and openly be driven by the use cases. Where something was popular (in majority of cases, or in this case, all of the use cases) it ought to taken as something important to our users?

RKrahl, 2018-11-04:
I agree that we need to discuss the criteria for adding properties and for classifying them as mandatory, recommended, or optional. That is why I made a proposal to formulate those criteria. @louatbodc, would you mind commenting my proposal in #7?

RKrahl, 2019-03-06:
Close as discussed in today's meeting.

closed 2019-03-06 by RKrahl