rdawg-pidinst/schema #8: add link to code list

oggioniale, 2018-11-04:
for the properties whose content refers to codelist, indicate the link to the codelist if they exist (e.g. for relatedIdentifierType a complete codelist could be http://pid.geoscience.gov.au/def/voc/ga/associationtype)

RKrahl, 2019-01-29:
I suggest that we must discuss the content, e.g. agree on the terms that should be in the controlled list of values for our schema first, before we can go and find an external resource for them. The best external resource does not help if the content does not fit our purpose.

See also the comment in #9.

markusstocker, 2019-02-07:
I agree with @RKrahl that we cannot simply take an existing code list without agreeing it is relevant for this schema.

closed 2019-02-07 by markusstocker