rdawg-pidinst/schema #13: manufacturerIdentifier

huberrob, 2019-02-22:
The term manufacturerIdentifier could be misunderstood it has to be clarified what it means:

1) The identifier given to and instrument by its manufacturer 
or
2) The identifier of the manufacturer itself.

RKrahl, 2019-02-26:
It is intended to mean an identifier of the manufacturer itself. The inclusion of this subproperty has been proposed by me in #3. What would you suggest to avoid misunderstandings?

markusstocker, 2019-02-26:
I agree it is ambiguous. The definition "Persistent identifier of the manufacturer" is clear, I think, but the question is whether this should be made more explicit in the attribute name, e.g. manufacturerInstrumentIdentifier or instrumentIdentifierByManufacturer.

RKrahl, 2019-02-27:
Well, if “Persistent identifier of the manufacturer” is clear, that is the current definition that we have in the schema. So either we assume that people take notice of the definition, then there should be no problem here, or we assume they don't and try to squeeze the definition into the property names. That would become really cumbersome for many properties. And then, if we assume that people do not take notice of the definition in the first place, why should we assume that they try to observe property names?

huberrob, 2019-02-27:
OK I see .. as a subproperty it is clear what is meant with it .. but the definition should be updated to avoid ambiguities.

RKrahl, 2019-02-27:
The current definition is “Persistent identifier of the manufacturer”. What other definition do you suggest?

huberrob, 2019-02-27:
Maybe: 'Persistent institution identifier which is used to identify the
manufacturer of the instrument or sensor, this can be an ORCID, ISNI or
similar'

-- 
Dr. Robert Huber,

PANGAEA - www.pangaea.de
_____________________________________________
MARUM - Center for Marine Environmental Sciences
University Bremen
Leobener Strasse
POB 330 440
28359 Bremen
Phone ++49 421 218-65593, Fax ++49 421 218-65505
e-mail rhuber@uni-bremen.de

RKrahl, 2019-02-27:
Some minor nitpicking:
- As far as I know, the scope of ORCID is strictly limited to people, so I believe there can't be an ORCID id for an instrument manufacturer. Better no examples then wrong examples.
- We use either “instrument” or “instrument instance” throughout the schema. No need to point out in the definition of `manufacturerIdentifier` that an instrument may be a sensor.
- I'm not sure if the identifier type must always be a specific “institution identifier”. I guess an URL, URN, or Handle could also be acceptable. So I suggest to write just “Persistent identifier which …”

In short, I'd suggest: “Persistent identifier which is used to identify the manufacturer of the instrument.” Ok?

markusstocker, 2019-02-27:
A couple of (shorter) alternatives:

Persistent identifier used to identify the manufacturer of the instrument

or

Persistent identifier that identifies the manufacturer of the instrument

or

Persistent identifier of the manufacturer of the instrument

RKrahl, 2019-03-05:
The third suggestion is (almost) what we have now: “Persistent identifier of the manufacturer”.

louatbodc, 2019-03-20:
Excellent. Happy to close this

louatbodc, 2019-03-20:
Whoops commented on the wrong issue! This is not the one I want to close

huberrob, 2019-03-21:
I would vote for : Persistent identifier used to identify the manufacturer of the instrument

RKrahl, 2021-02-26:
Note that  `manufacturerIdentifier` is a subproperty of `Manufacturer`.  I suggest that “of the instrument” is redundant with the overall context of this schema, but also with the description of `Manufacturer`: “The instrument's manufacturer(s) or developer. […]”. Therefore, I'd propose: “Persistent identifier used to identify the manufacturer”

PR #42 (2021-03-10): Amend the description of ownerIdentifier, manufacturerIdentifier, and modelIdentifier
In #13 it was observed that the description of `manufacturerIdentifier` might be ambiguous whether this means to be the identifier given to the instrument by its manufacturer or the identifier of the manufacturer itself.  This PR changes the description to “Persistent identifier used to identify the manufacturer” to make this clear. It also changes the description of `ownerIdentifier` and `modelIdentifier` accordingly. It thus closes #13.
closed 2021-03-10 by RKrahl