rdawg-pidinst/schema #20: Ensure basic compatibility with TEDS /  IEEE 1451.4 (plug & play sensors)

huberrob, 2019-03-07:
IEEE 1451.4 is a industry standards for plug & play sensors and TEDS stands for Transducer Electronic Data Sheets

TEDS provides ways to identify sensors, basic TEDS includes:
• Manufacturer ID
• Model Number
• Version Letter
• Version Number
• Serial Number

https://standards.ieee.org/content/dam/ieee-standards/standards/web/documents/tutorials/teds.pdf

RKrahl, 2019-03-07:
Could you elaborate in more detail, why compatibility is an issue here and what you suggest to do to ensure this compatibility?

huberrob, 2019-03-07:
In terms of metadata compatibility e.g. to allow semi-automatic entry of
our PIDINST schema in the future by plug & play sensors

-- 
Dr. Robert Huber,

PANGAEA - www.pangaea.de
_____________________________________________
MARUM - Center for Marine Environmental Sciences
University Bremen
Leobener Strasse
POB 330 440
28359 Bremen
Phone ++49 421 218-65593, Fax ++49 421 218-65505
e-mail rhuber@uni-bremen.de

louatbodc, 2019-03-20:
Hi Rolf and Robert

I guess the marine domain is one of the further advanced use cases using sensor-machine readable technology/standards, particularly with plug 'n' play sensors. This is probably why some of us have been active in including some of the more sensor specific properties and that they are understood by machine agents. However, I also appreciate there needs to be a happy minimum that satisfies all use cases which may be why we've not been able to include all.

I think manufacturer, model and serial no. are now better described for this purpose - although I do think we need a controlled list of values for alternateidentifiertype, so a machine can understand they are looking at serial no. or an inventory number. From our use cases, we've only identified two values so far so this should be practical. Should there be a need for others, we can always add to it. Should it rapidly expand we can always revise. If we include the use of 'other' (as DataCIte do) this will also help for now.

A lot of manufacturers will include the version in the model name - is this a happy medium Robert? Or do you think we need a version property? 

Cheers
Lou

P.S. We could publish controlled list of values at BODC on our NERC Vocabulary Server. They will have full provenance (resolvable URIs to a machine readable description (e.g. http://vocab.nerc.ac.uk/collection/L22/current/TOOL0037/)).

RKrahl, 2021-02-26:
I guess this has been settled by now. I suggest to close.

RKrahl, 2021-03-10:
Close as discussed in to meeting today.

closed 2021-03-10 by RKrahl