rdawg-pidinst/schema #21: Controlled list of values for dateType

RKrahl, 2019-03-20:
We still do not have finalized the controlled lists of values for several properties. I hereby open individual issues for each of them.

For `dateType` we currently have:
+ `Commissioned` and
+ `DeCommissioned`.

This may be used to indicate the period of time that the instrument is / has been active.  Alternatively to these two one could think of a value `InOperation` that could be set to a date interval rather then a single date, e.g. something like `<date dateType="InOperation">2015-09-18--2017-12-31</date>`.

What other dates need to be recorded in the PID? Should we have a way to record dates of modifications of the instrument or should we assume that always a new PID is attributed to the instrument after each modification?

huberrob, 2019-03-21:
These types rather seem to describe something like an 'event' not a date. Id prefer names like:

commissionDate
decommissionDate

More types:

purchaseDate
accessionDate
loanDate
returnDate
calibrationDate

Maybe should also think about to change the schema here and use e.g. 'Event' instead of 'Date' as main element which has a 'eventDate' 'eventType' subproperty..

markusstocker, 2019-03-21:
Do we really want to push information about things like calibration into this schema? Without being an expert on instrument calibration, to my understanding this information is (at least for some instruments) rather dynamic. Do we expect people to actually update this at the PID infrastructure level?

In other words, is there a use case for applications that need calibration dates at the global level of PID infrastructure, for all registered instruments?

I can of course see the appeal of having as much of this information harmonized in a central index but I worry about synchronization.

Or is the suggestion to issue a new PID each time something changes with this data?

huberrob, 2019-03-21:
I agree this is not really necessary for registration purposes, so just
ignore those from the list ;) Same for comissining and decomissioning dates
these also hard to sync..

Schema.org https://schema.org/Product has productionDate and purchaseDate,
maybe these would be sufficient?

-- 
Dr. Robert Huber,

PANGAEA - www.pangaea.de
_____________________________________________
MARUM - Center for Marine Environmental Sciences
University Bremen
Leobener Strasse
POB 330 440
28359 Bremen
Phone ++49 421 218-65593, Fax ++49 421 218-65505
e-mail rhuber@uni-bremen.de

RKrahl, 2021-07-07:
Suggestion: leave it with `Commissioned` and `DeCommissioned` for version 1.0.

ErinKenna, 2021-07-07:
I'd agree with the comments about keeping things like calibration out. 
I've been creating separate records for calibrations, maintenance/service and deployments and they have an attribute of InstrumentID (or PlatformID which might host multiple instruments). If I need to construct an instrument record I'd interrogate all the other records for the InstrumentID to build a history. If I implement an API then responses for an instrument request might include a list of calibration and deployment identifiers as a convenience but they wouldn't exist in the actual instrument record in the backend. 

`purchaseDate` (is this `Commissioned`?) and `decommissionDate`* are probably the main ones I need. An extra date I might add to the actual instrument record might be a date for when the next maintenance is due/scheduled. 

I haven't been handling loan/return types of dates but when I do I was expecting to create separate records for that sort of thing as well with perhaps just the current status (on loan, available, ...) being updated in the instrument record as a convenience. 

*I'm reading `decommissionDate` as the end of the life of the instrument. Not sure if that's right? Is there somewhere where I can see the vocabulary and the definition of the terms?

huberrob, 2021-07-07:
Agree, but I'd prefer to rename _Commissioned_ into _commissionedDate_  in order to make clear that this is not  Boolean ;)

ErinKenna, 2021-07-07:
Just looking at the recommendation in the white paper <https://rda-pidinst.readthedocs.io/en/latest/white-paper/create-new-pid.html> and the 'use of the relatedIdentifier property to represent superseded PID records'.  
I'm wondering if something complementary might be required here like a 'supersededDate'? 
This might give a clearer picture of when the record was superseded. And infer a _kind of_ end-of-life date for this record as an alternative to `decommissionDate` since that would now be captured in the superseding record now presumably.

RKrahl, 2021-07-23:
I disagree on _commissionedDate_. `dateType` is a subproperty of `Date`. The serialization in XML would look like
```xml
<date dateType="Commissioned">2019-12-31</date>
```
From that context it is obvious that the value of `dateType` describes the nature of that date. There is no possibility of confusion with a boolean value. Appending "Date" to the value would be redundant and useless.

RKrahl, 2021-07-23:
@ErinKenna, the `Commissioned` / `DeCommissioned` pair of dates describes the period of time when the instrument that this PID record describes was in active.

What you cite from the White Paper refers to the situation that an instrument has been modified to an extent that you would not consider it the same instrument any more and thus you create a new PID for the modified instrument. In a sense, that means that the old instrument that the previous PID was referring to ceased to exist and a new one referred to by a new the PID popped into existence. In that situation, I'd consider it most appropriate to tag the date of the modification as `dateType=DeCommissioned` in the old PID record and as `dateType=Commissioned` in the new one. I don't see the need to add a new `dateType` for this kind of event.

RKrahl, 2021-07-30:
In the preparation of submitting the schema as a RDA recommendation, we plan to get a decision on all pending open questions during the next monthly meeting on 4th August.

For the controlled list of values, I suggest to leave it with those that have already been used in the prototype implementation or that have a clear rationale for a use case.  It will be much easier to add another missing value in a future version of the schema than to remove one that turn out to be less useful.

For `dateType`, I suggest to leave it with `Commissioned` and `DeCommissioned` for version 1.0.

closed 2021-08-05 by RKrahl

ErinKenna, 2021-08-05:
> @ErinKenna, the `Commissioned` / `DeCommissioned` pair of dates describes the period of time when the instrument that this PID record describes was active.
> 

Thanks for the clarification @RKrahl

PR #52 (2021-08-05): Controlled list of values for dateType
Set controlled list of values for `dateType` to `Commissioned` and `DeCommissioned`.  Close #21.