rdawg-pidinst/schema #22: Controlled list of values for relatedIdentifierType

RKrahl, 2019-03-20:
We still do not have finalized the controlled lists of values for several properties. I hereby open individual issues for each of them.

RelatedIdentifier is the identifier of a releted entity. The subproperty relatedIdentifierType should be the type of the PID that has is used here. So essentially need an enumeration of all possible PID types. At the moment we have:

+ `PIDINST`,
+ `DOI`,
+ `Handle`,
+ `URL`,
+ `URN`.

This list is certainly not yet complete. We could just copy the list for `relatedIdentifierType` from [DataCite](https://schema.datacite.org/meta/kernel-4.1/). Is that list complete for our purpose?

RKrahl, 2021-07-07:
Suggestion: take DataCite list as a basis, but review it.

jklump, 2021-07-07:
RRID should be included in the list. LSID is no longer active.

RKrahl, 2021-07-22:
I guess, funding reference is more complicated, because you would need to reference a grant from a particular funder.  I'm not aware of any established identifier that would reference both at the same time.  DataCite has a dedicated complex property `FundingReference` with subproperties `funderName`, `funderIdentifier`, `awardNumber`, `awardURI`, and `awardTitle` for this.

So I'd suggest to leave it out here for the time being and discuss it in a separate issue if needed.

PR #50 (2021-07-23): Review controlled list of values for relatedIdentifierType
Take `relatedIdentifierType` from DataCite 4.4, omitting LSID and adding RRID and RAiD.  Resolve #22.
closed 2021-07-23 by markusstocker