rdawg-pidinst/schema #25: DataCITE mapping feedback

louatbodc, 2019-04-25:
Hi Rolf

Thanks for doing this. It certainly highlights some issues as you have pointed out. As you say, some of the DataCite properties do seem a bit tenuous for describing instruments and seem to need a lot of explanation for an end-user to apply appropriately. While I understand the DataCite schema is meant to be flexible (to allow it to be used across domains and applications), is there a risk this could also lead to different variations, reducing interoperability and discoverability? 

I can see there are instruments already in DataCite using some of these properties with resourceGeneralType = 'Physical object' so it is possible to use the DataCite properties. Though based on the XML, I do sometimes struggle to know if I'm looking at an instrument instance or an identifier for an instrument type/model in an inventory[1,2]. This is why it would be good to have a more specific resourceGeneralType and to include some of the more specific instrument properties (e.g. ModelNumber, SerialNumber, dates, owner etc.)

Cheers
Lou

[1] https://search.datacite.org/works/10.7282/t3pr7z04
[2] https://search.datacite.org/works/10.7282/t3nv9mcd

markusstocker, 2019-04-29:
Hi Rolf, thanks, indeed. I reviewed the material and I agree with your observations. I also think that Lou is right that schema flexibility comes at the cost of interoperability.

RKrahl, 2019-04-29:
Hi @louatbodc: concerning the interoperability issue: that is exactly why I suggest that we would need extended documentation on instrument PIDs in the DataCite schema with clear guidelines.  I do hope that such guidelines could streamline the creation of the metadata to an extend that interoperability will be provided in practice.

RKrahl, 2021-02-26:
I guess this has been settled by now. I suggest to close.

RKrahl, 2021-03-10:
Close as discussed in to meeting today.

closed 2021-03-10 by RKrahl