rdawg-pidinst/schema #28: Model Name

tedhabermann, 2019-04-29:
Using AlternativeIdentifier for model name makes sense to me, with a type = model. I suspect that people use model names in sentences like "I need a X754". I think this is an identifier granularity issue. 

Is it possible to have a specific identifier for a thing as well as other identifiers at other levels of granularity? I think so. I'm not sure how one searches alternateIdentifier types in DataCite.

RKrahl, 2019-04-29:
Yes, exactly, people might say "I need a X754". But an identifier is something that people would use with a definite article "I generated this data using _the_ E2 at BER II". If people tend to use an indefinite article, then this may already be taken as a hint that it is unsuitable as an identifier.

RKrahl, 2019-04-29:
… continued:

So using either `AlternativeIdentifier` or `RelatedIdentifier` to point to the model would require to have real identifiers on the model level.  It would certainly be nice if such identifiers would exist, but it's not in the scope of this group to create them. Model names are managed by the manufacturers and from their point of view they are mostly a marketing tool.  So I doubt they would be happy if anybody would try to interfere with their naming schemes or that it would be feasible to coerce them to use model names in any consistent manner, let alone to mint proper identifiers on them.

RKrahl, 2021-02-26:
Meanwhile, we have `Model` with sub properties `modelName`, `modelIdentifier`, and `modelIdentifierType`. I guess this issue has been settled. I suggest to close it.

RKrahl, 2021-03-10:
Close as discussed in to meeting today.

closed 2021-03-10 by RKrahl