rdawg-pidinst/schema #30: resourceTypeGeneral

tedhabermann, 2019-04-29:
It would be desirable to add "Instrument" to the controlled list of values for resourceTypeGeneral.

We suggested platform, instrument, or sensor for resourceType. If we add instrument to resourceType do we use it for all three? Maybe "device" would accomplish the goal but also be general?

RKrahl, 2019-04-29:
I would say, yes, `resourceTypeGeneral=Instrument` would fit for all three `resourceType` values. I'd consider "device" to be to unspecific.

markusstocker, 2019-04-29:
Talking about being specific, IMO we need to clarify whether we should speak of "MeasuringInstrument". There may be a time when clarinets are identified by DOI.

RKrahl, 2019-04-29:
"MeasuringInstrument" sounds too complicated for my guts feeling.

Also I don't think there is a need to exclude other, not-measuring instruments here. I agree to limit the scope of the group to measuring instruments, e.g. we will only consider those when we define the PID infrastructure and formulate the metadata schema. But if later on somebody come along and find the result to be useful for other types of instruments as well, why not, it does not harm if the result  is used for other purposes as well.

RKrahl, 2021-02-26:
This is now discussed in datacite/schema#70. I suggest to close here.

RKrahl, 2021-03-10:
Close as discussed in to meeting today.

closed 2021-03-10 by RKrahl