rdawg-pidinst/schema #32: Metadata Schema - Elements

kitchenprinzessin3880, 2019-08-12:
Identifier
-- 'Unique string that ..' - does sit to be only string?
-- 'Allowed values, constraints, remarks' - what is 'PIDINST'? in the context of this metadata element?

RKrahl, 2019-08-27:
Some comments:

- Ad 1. Identifier:
  An identifier is an opaque bit of information that has no semantic meaning at all by itself other then uniquely identifying the entity it represents. It is commonly stored as a string, What else do you have in mind?
  The name `PIDINST` is used in the schema merely as a placeholder for the type of instrument PID while there is no final decision on the technical platform to be used. If we go for DataCite DOIs, it stands for `DOI`, if we go for ePIC, it stands for `Handle`.

- Ad 2. Description:
  You ask for another attribute to link external technical specification? We already have that in the schema: `RelatedIdentifier` with `relationType=IsDescribedBy`.

- Ad 3. Controlled list of values:
  I'm not sure if I completely understand your suggestion, you mean we should use some type registry for the attributes having controlled lists of values and reference them by the type PID issued by the registry? Sure, we could do that and this approach certainly has its merits. But it would require a significant amount of extra work and I don't believe we have the resources to do that in this working group.

- Ad 4. location:
  This has already been discussed in #17. The conclusion was that there are many unresolved issues with that. It has been suggested to rather link deployments of an instrument externally using `RelatedIdentifier` (needs the definition of a suitable `relationType`, though).

kitchenprinzessin3880, 2019-08-28:
Hi Rolf,
1. Identifer - 'string' - i want to confirm if it only allows [a-zA-Z] or alphanumerics (\w)
2. RelatedIdentifier with relationType=IsDescribedBy --> noted. 
3. significant amount of extra work --> please provide definition of the enumerations. 
4. location -> noted.

RKrahl, 2021-02-26:
I submitted #44 that hopefully clarifies the value for `identifierType`. Otherwise I assume that this issue has been settled by now. I suggest to close.

RKrahl, 2021-03-10:
Close as discussed in to meeting today.

closed 2021-03-10 by RKrahl