rdawg-pidinst/schema #33: Mandatory Manufacture info

huberrob, 2019-09-04:
Manufacturer and manufacturerName now are mandatory in our schema. However this information is not always available. Further, it is not mandatory in SensorML 2.0 so there is a compatibility issue.

Possible Solutions:
1) Make Manufacturer info Optional
2) Provide a default entry like 'Not Given' or 'Unknown' (but it has to be discussed if this is confusing in comparison to solution 1)

RKrahl, 2019-09-04:
I would suggest the second option. In fact, I would suggest to adopt the “Standard values for unknown information” from DataCite (see Appendix 3 in the [DataCite 4.3 Metadata Schema Documentation](https://schema.datacite.org/meta/kernel-4.3/)) that may be used for any property if the value is not available for any reason.

These values include:

value   | meaning
------- | -------------------------------------
`:unal` | unallowed, supressed intentionally
`:unap` | not applicable, makes no sense
`:unas` | value unassigned (e.g., Untitled)
`:unav` | value unavailable, possibly unknown
`:unkn` | known to be unknown (e.g., Anonymous)
`:none` | never had a value, never will
`:tba`  | to be assigned or announced later

and others.

This approach has the advantage that the issuer of the PID has at least the obligation to state that this value is not known and is not just forgotten to be filled in. And it also gives the opportunity to state why this value is not given.

amacario, 2019-09-04:
Thanks for the suggestion.  Technically no big deal to implement a couple of the values above as option in the frontend but hard to get the people to fill out this property later on if we set to "tba" all the mandatroy properties which are currently missing a value.   The human factor again...

Ditto for property "instrument model"

amacario, 2019-09-04:
And we have the issue that some of the sensors were indeed built in AWI's  "Werkstatt" so that they do not have a registered manufacturer and model name.

RKrahl, 2019-09-04:
The latter is not a problem.  If the sensor is built by AWI's "Werkstatt", the manufacturer is AWI. We explicitly state in the definition of the Manufacturer property: “This may also be the owner for custom build instruments.”

RKrahl, 2021-02-26:
I guess this has been settled by now. I suggest to close.

huberrob, 2021-03-01:
I agree with you, this can be closed..

-- 
Dr. Robert Huber,

PANGAEA - www.pangaea.de
_____________________________________________
MARUM - Center for Marine Environmental Sciences
University Bremen
Leobener Strasse
POB 330 440
28359 Bremen
Phone ++49 421 218-65593, Fax ++49 421 218-65505
e-mail rhuber@uni-bremen.de

closed 2021-03-08 by RKrahl