rdawg-pidinst/schema #34: rename  'VariableMeasured' to 'MeasuredVariable'

uscw, 2019-11-12:
Because 'VariableMeasured' is 0-n, a kind of array needs to be defined in JSON as well as in XML, and this array needs a name. The usual naming convention here would lead to 'VariableMeasureds', which is nonsense. The suggestion to use 'MeasuredVariables' is well phrased and would be a semantically correct mapping, but it needs a semantical understanding to derive it this way, nothing for automated processes. Furthermore it is in contradiction to the naming convention as far as understand it. My suggestion would be to rename 'VariableMeasured' to 'MeasuredVariable', which make sense anyway and leads automatically to the correct name for the array.

markusstocker, 2019-11-12:
Just as a side note here, this may have been inspired by http://schema.org/variableMeasured, used in Dataset. I presume a measuring instrument can actually measure multiple variables but my understanding of instrumentation is very limited. I guess it is also a granularity issue. It may often be so that what appears to be an instrument that measures multiple variables is actually a system of instruments each measuring only one variable. If we do need an array, then I surely agree with the name change (MeasuredVariables makes more sense, though it could be VariablesMeasured). My bigger issue is whether a measuring instrument actually measures more than one thing.

RKrahl, 2019-11-13:
I would call a thing that measures only one single variable a sensor. It might be appealing from a very systematic and strict point of view to attribute every single sensor its own PID. But I would guess that such an approach would be far too detailed for most practical use cases. And even then, we would still need a mean to identify those bigger things that we have sitting in our labs that are in general complex devices composed of many individual sensors and that I would call an instrument.  We deliberately did not define in detail what an instrument is in the scope of our WG and left the granularity issue to the discretion of the users that issue the PIDs. Consequently, the schema must at least be able to accommodate the case that an instrument measures many different variables.

@uscw: thanks for bringing this up! Your argument is sensible. I support the suggestion to rename this property.

commit 30533f74 (2019-12-05): Rename 'VariableMeasured' to 'MeasuredVariable'.  Close #34.
closed 2019-12-05 by RKrahl