rdawg-pidinst/schema #39: InstrumentType identifiers

louatbodc, 2020-05-06:
in addition to controlled identifiers for model names such as RRIDs, there are standardised terms for instrument types. At the moment I can only apply the standardised term as the value. 

E.g. 
InstrumentType = ‘http://vocab.nerc.ac.uk/collection/L05/current/LAB07/’;

This is not very human friendly or good for machines to understand. Thus it would be good have attributes associated to instrument type that help clarify a human-readable value with identifiers for any domain .e.g.

	InstrumentType = ‘electron microscopes’; (free text)
	InstrumentTypeIdentifier = ‘http://vocab.nerc.ac.uk/collection/L05/current/LAB07/’; (identifier)[optional]
	InstrumentTypeIdentifierType = ‘URL’; (e.g. URL, URN, DOI, Handle) [mandatory if using InstrumentTypeIdentifier]

	It would even be good to have a scheme option in general for several properties:

	InstrumentTypeIdentifierScheme = ‘rdf+xml’; [optional. Only used if InstrumentTypeIdentifierType is specified]

louatbodc, 2020-05-06:
scheme should be freetext

RKrahl, 2021-07-22:
@louatbodc, you suggested to also add `instrumentTypeIdentifierScheme`. We don't have a `*IdentifierScheme` subproperty in other  `*Identifier` properties by now. I agree it would make sense. But for the sake of consistency I suggest to add a `*IdentifierScheme` subproperty to all  `*Identifier` properties then.

I will therefore split this into two pull requests: one to add `instrumentTypeIdentifier` and another one to add a scheme subproperty to all identifiers.

RKrahl, 2021-07-22:
@louatbodc, after closer consideration, I start to doubt about the `*IdentifierScheme` subproperties. What would actually be the information that you would put there? What exactly would you suggest should be the definition?

PR #48 (2021-08-05): Add instrument type identifier
- Change `InstrumentType` to a complex property having subproperties `instrumentTypeName`, `instrumentTypeIdentifier`, and `instrumentTypeIdentifierType`.
- Change the cardinality of `InstrumentType` to 0-n.

Resolve #39.
closed 2021-08-05 by RKrahl