rdawg-pidinst/schema #58: Instrument Status

kitchenprinzessin3880, 2021-11-03:
At present the schema captures the operation status of an instrument through dates (commissioned and decommissioned). Suggest including an explicit metadata property to represent instrument status (active, inactive).

RKrahl, 2021-11-03:
I guess it should have a controlled list of values.  You already suggested `Active` and `InActive`.  I guess we would at least also need `Unknown`.  Then the question is whether we want to distinguish between temporarily and definitely inactive.

Another question would be the obligation (mandatory, recommended, optional) of that property. I suggest it should not be mandatory, as I assume many organization would prefer not to maintain the status information in the PID record.

kitchenprinzessin3880, 2021-11-03:
+1 recommended property

RKrahl, 2021-11-03:
Or even optional. But this will be subject to discussion.