rdawg-pidinst/schema #59: Cite the project?

EstherPlomp, 2021-11-04:
Hi all! I wanted to mention this schema in an editorial I'm writing. How should I refer to this? Do you have a version available on Zenodo? Or a report/summary/output?

Thanks!

RKrahl, 2021-11-04:
@EstherPlomp you are a little bit too early. Right now, we are preparing to submit the schema to community review as an RDA recommendation. If it gets endorsed, it will obviously be published, I suggest on Zenodo.

EstherPlomp, 2021-11-14:
Thanks for the quick response! Good to hear that there are plans to share it through Zenodo (or the RDA), I hope that will still be in time for my editorial and otherwise I'll refer to the working group.

RKrahl, 2021-11-14:
Just to give you a time horizon: we hope to get the submission done in December.  After that, it needs to go through the RDA review process that I don't know how long it will take.  In the meanwhile, you could cite is as something like: "RDA Persistent Identification of Instruments WG (2021). Metadata Schema for the Persistent Identification of Instruments. To be submitted as a RDA recommendation." (The final version will name the individual authors, but we still need to get confirmation from each of them.  So at the current stage, I guess it's better just to put the WG as author.)

See the [current draft](https://drive.google.com/file/d/1oN4_efdNthjHNVkjwfgyUIhqg2626S9C/view?usp=sharing) for the document.

markusstocker, 2021-11-14:
@EstherPlomp There is also this you can use to cite/credit the work:

Markus Stocker, Louise Darroch, Rolf Krahl, Ted Habermann, Anusuriya Devaraju, Ulrich Schwardmann, Claudio D’Onofrio, and Ingemar Häggström (2020). Persistent Identification of Instruments. Data Science Journal, 19:18. [https://doi.org/10.5334/dsj-2020-018](https://doi.org/10.5334/dsj-2020-018)

EstherPlomp, 2021-11-14:
Thanks @markusstocker! I'll refer to that one for now then, as I expect this editorial to go out earlier than the RDA Recommendation will be available!

RKrahl, 2022-01-12:
Just for the records: the schema is now submitted to RDA as a recommendation and got a DOI.  So the citation would be for now:

Krahl, R., Darroch, L., Huber, R., Devaraju, A., Klump, J., Habermann, T., Stocker, M., & The Research Data Alliance Persistent Identification of Instruments Working Group members. (2021). Metadata Schema for the Persistent Identification of Instruments. Research Data Alliance. https://doi.org/10.15497/RDA00070

It's still in RDA community review and not yet endorsed and not yet pushed to Zenodo.  But I suggest the citation line will be mostly stable.  Maybe we'll add a version number 1.0 to the final version.

PR #65 (2022-04-06): Update README: now formally endorsed as a RDA recommendation
Update README:

- the schema it is now formally endorsed as a RDA recommendation,
- add a citation hint,
- add a DOI badge.

Close #59.
closed 2022-04-06 by RKrahl