rdawg-pidinst/schema #67: Experimental technique

RKrahl, 2023-03-09:
The [ExPaNDS](https://expands.eu/) project has developed [PaNET – The Photon and Neutron Experimental Techniques Ontology](https://github.com/ExPaNDS-eu/ExPaNDS-experimental-techniques-ontology). I assume other (experiment performing) scientific communities might have similar vocabularies for their techniques. Obviously, it would be interesting to link an instrument with the experimental technique(s) it has been designed to use. Surprisingly enough, there doesn't seem to be a proper place in our schema to do that.

Experimental techniques seem to be somewhat complementary to `MeasuredVariable`: the latter is the physical observable that the instrument records. The technique goes way beyond that, it describes the physical phenomenon that the instrument makes use of or the measurement protocol it performs in order to make an observation. `MeasuredVariable` is fair enough for simple instruments like temperature sensors. But it is too limited to describe complex instruments such as an experimental station for x-ray absorption spectroscopy.

Interestingly enough, it is possible to include the link from the instrument to the experimental technique in the DataCite schema: the `subject` property in DataCite is mostly used to just add keywords. But using its subproperties, it is versatile enough to also establish the link to a PaNET term. I did that excercise for one of our [instrument DOIs](https://doi.org/10.5442/NI000001) (see the [DataCite XML metadata](https://api.datacite.org/application/vnd.datacite.datacite+xml/10.5442/ni000001)):
```xml
  <subjects>
    <subject>BER II</subject>
    <subject>Diffraction</subject>
    <subject>Neutron source</subject>
    <subject schemeURI="http://purl.org/pan-science/PaNET/" subjectScheme="The Photon and Neutron Experimental Techniques Ontology" valueURI="http://purl.org/pan-science/PaNET/PaNET01217">neutron diffraction</subject>
  </subjects>
```

kara-mela, 2025-05-19:
At DESY, we are currently extending the metadata use at different steps of the workflows. We consider to use PIDINST and also the PaNET terms. It would be beneficial to establish a direct connection on the instrument level, as the set of provided techniques is quite static and rarely undergoes updates. 

The PaNET has created a document with recommendations on how to [include their terms as metadata in DataCite](https://doi.org/10.5281/zenodo.14191575), which basically corresponds to the previous suggestions (except for some details).

```
<subject xml:lang="en" subjectScheme=”PaNET”
  schemeURI=”http://purl.org/pan-science/PaNET/PaNET.owl”
  valueURI=”http://purl.org/pan-science/PaNET/PaNET01135”
  classificationCode=”PaNET01135”
  >absorption spectroscopy</subject>
```

It would be great to have the `subject` property available, before we intensively start working with PIDINST, such that we can include the experimental terms right from the beginning.

RKrahl, 2026-02-04:
I agree that we should not confuse this property with keywords or subjects.  Maybe my initial comment was misleading: I meant to say: “look what's posible in DataCite, we want that as well”. I didn't meant to suggest copying the DataCite `Subject` property over to PIDINST.  This proposal is not to be confused with #72.

My proposal was to have a property for the technique (lets use that neutral term for the moment) the instruments implements.  I would define the technique as: 

  “the protocol that the instrument follows or the physical phenomenon it makes use of in order make its observation.”

I believe this is significant enough for the description of an instrument to merit a dedicated property in the schema.  And yes, this would be a multi valued property, an instrument may implement more than one technique.  And of course, the property should allow and encourage linking to existing controlled vocabularies, so we'd need subproperties for adding the PID of the term in that vocabulary.

I don't think `ScientificDomain` would be appropriate here.  This is really something different.

Concerning the name of the property, I don't stick to `ExperimentalTechnique`.  I'm happy with `MeasurementTechnique`.  I'm not so sure about ` InstrumentTechnique`, because the technique exists independently of a given instrument and might be implemented in different ways.

RKrahl, 2026-02-04:
NB: one argument in favour of `MeasurementTechnique` might be that this name hints that this property is complementary to `MeasuredVariable`.